GyverLibs/microDS18B20 #1: some questions about multithread.

youxiaojie, 2021-07-04:
1.
int  DS_rawToInt ( uint16_t data); // convert raw data to int temperature
float  DS_rawToFloat ( uint16_t data); // convert raw data to float temperature

and
 DS_TEMP_TYPE calcRaw ( uint16_t data) {
 return ((DS_TEMP_TYPE) data / 16 ); // Calculate the temperature value
 }

are the same, why define two funtion?
 
2. if I want use multithread, shoud I change delay() to millis()-startime>= period format?
3. is your code must cooperate with your microwire lib?
4. did your two libs take less resources than officials?
thanks!

GyverLibs, 2021-07-04:
Hello! 
1. DS_rawToInt/DS_rawToFloat are outside of class, so you can use it for convert raw data from transmission from other device
2. Of course, good code should not contain delays() at all! You can use [TimerMS ](should not contain) library or [GyverOS ](https://github.com/GyverLibs/GyverOS)task manager
3. No, microWire is I2C, this library is OneWire interface
4. All my libs are highly optimized and take less resources, especially "micro". microDS18b20 can be running on Attiny13 and take just 300 kB of Flash

youxiaojie, 2021-07-04:
1.sorry, misspell, microOneWire.h, if I use your microds18b20, I don't need official onewire lib.
2. I see your two .h files contain delay(), If I cooperate with taskscheduler or gyveros, need I rewrite two .h to change delay() to millis()-startime>= period format.

`All my libs are highly optimized and take less resources, especially "micro". microDS18b20 can be running on Attiny13 and take just 300 kB of Flash, while Dallas library takes about 2.5 kB Flash and almost all SRAM memory =)??`
300kB? a typo?

GyverLibs, 2021-07-04:
1. Of course, you dont need official lib, it is not included nowhere in code
2. Delays in library provide one wire interface, you cant rid of them because of timing. Communication by one wire takes a little time, dont worry about it
3. Yes, typo. 300 Bytes of course i mean

closed 2021-07-27 by GyverLibs