GyverLibs/microDS18B20 #2: Поддержка esp32

Phantasmagory, 2021-09-07:
До обновления вроде работала с esp32. Возможно прикрутить поддержку?

GyverLibs, 2021-09-07:
Нужны конкретные симптомы, нет возможности потестить

Phantasmagory, 2021-09-07:
Компилируется. На выходе 0,00 градусов.

GyverLibs, 2021-09-08:
при каком способе опроса? Какой пример загружен? Пин правильно указан? Это ведь esp, там всё по другому

GyverLibs, 2021-09-08:
проверил на esp8266 - всё работает во всех режимах

Phantasmagory, 2021-09-08:
Загружал разные примеры и адрес считать пытался. Пины: пробовал и номера и gpio и Dpin(D не понимает). 
Тут тоже пробовал разные вариации:
#define DS_TEMP_TYPE float  	  // float/int - тип данных для рассчета температуры float / int - (экономит flash, выполняе
#define DS_CHECK_CRC true		  // true/false - проверка контрольной суммы принятых данных - надежнее, но тратит не
#define DS_CRC_USE_TABLE false    // true/false - использовать готовую таблицу контрольной суммы - значительно быстрее, 

Со старыми версиями тоже не работает, то я перепутал с есп8266.
В старой версии я комментировал SREG в microOneWire и работало с есп8266.

GyverLibs, 2021-09-08:
а если попробовать другие пины? У 32-ой вроде с этим какие то проблемы

eviltony90, 2021-09-21:
Пробовал с ESP32 на разных пинах и разных версиях плат в Arduino IDE (1.0.6 и 2.0.0). Не работает - пишет все время 0 градусов C. При этом стандартная библиотека работает на версии плат 1.0.6., а на 2.0.0 не компилируется вообще.

GyverLibs, 2021-09-22:
Это очень странно, интерфейс реализован тупо digitalRead/write и delay, оно не может не работать) попробую запрячь автора низкоуровневой части библиотеки, закажет себе 32-ю и может починит

Basoil, 2022-01-24:
Подтверждаю - на ESP32 не работает. Другие библиотеки (DallasTemperature) работают.
Версия библиотеки - 3.6
Версия ядра ESP32 - 2.0.2
Перепробовал подключение на все доступные пины - не работает.

GyverLibs, 2022-01-24:
Жалко, у меня нет железки чтобы отладить

GyverLibs, 2022-02-18:
докладываю: коллеги-индусы задефайнили в ядре ESP32 константы пинмода на другие цифры, "не как у всех", поэтому сигнал с пина не поступал. Актуальная версия протестирована на esp32 и отлично работает, поздравляю всех неравнодушных =)