GyverLibs/microDS18B20 #5: На ESP8266 вместо отрицательной температуры возвращает +95.3

Basoil, 2021-11-02:
Температуры выше +3 возвращает нормально.
при t<+3 - возвращает +95.. +96
Использован пример   one_pin_one_sensor.ino
Датчик - герметичный с проводом 50 см. С Али.

GyverLibs, 2021-11-04:
Какая версия библиотеки? Отрицательные температуры пофиксили около месяца назад

Basoil, 2021-11-04:
> Какая версия библиотеки? Отрицательные температуры пофиксили около месяца назад

Версия 3.3

Есть еще одна поделка, на Atmega328. Тоже отправляет +95..

Скетч такой:
`
#include <microDS18B20.h>                      // Датчик температуры  DS18B20
#include <GyverPower.h>                         // библиотека для снижения энергопотребления
#include <Oregon_TM.h>                         // библиотека эмуляции и передачи датчика Oregon

#define DS18B20_pin   2
#define RF_pin        4

MicroDS18B20<DS18B20_pin> sensor1;             
Oregon_TM transmitter(RF_pin, 19);

float shower_temp;

void setup() {
  Serial.begin(9600);                                 // включаем Serial
  pinMode(13,OUTPUT);
  transmitter.setType(THGN132);
  transmitter.setChannel(2);        				  // Номер канала для THGN132 - 1...3
  transmitter.setBatteryFlag(0);                      // Флаг разряженной батарейки
  transmitter.buffer_size = 19;
  
  power.autoCalibrate();
  power.setSystemPrescaler(PRESCALER_1);
  power.setSleepMode(POWERDOWN_SLEEP);
  power.hardwareDisable(PWR_ADC | PWR_TIMER1);
  power.bodInSleep(false);
}

void loop() {     
  sensor1.requestTemp();                              // читаем датчик температуры
  shower_temp = sensor1.getTemp();

  transmitter.setTemperature(shower_temp);
  transmitter.setHumidity(98);
  transmitter.setComfort(shower_temp, 98);            // Расчёт передаваемого индекса комфорта
  transmitter.SendPacket();                           // отправляем в эфир
  digitalWrite(13, HIGH);
  delay(1);
  digitalWrite(13, LOW);
  power.sleepDelay(30000);

}`

GyverLibs, 2021-11-04:
Очень интересно. А пример из библиотеки в таких же условиях адекватно работает?

Basoil, 2021-11-05:
Снял приборчик, проверил, что <Oregon_TM.h>  корректно передает отрицательные температуры,
Залил скетч с версией 3.4 -  и все нормально, минус передает.
Может у меня стояла версия библиотеки более ранняя <3.2? Хотя IDE при обновлении говорил, что стоит 3.3

GyverLibs, 2021-11-05:
Минус не работал в 3.1. Мне очень интересно, откуда взялось число 95))

Basoil, 2021-11-05:
Посмотрел по графику с Home Assistant - как температура упала до 0.0 - датчик стал возвращать 95.9, и дальше, с падением температуры на улице, тоже стал падать
[image]

GyverLibs, 2021-11-05:
Да, это похоже на версию 3-3.1, основной разработчик либы буковку пропустил)

Basoil, 2021-11-05:
Значит спишем на мою невнимательность )
Спасибо за либы! Пользуемся, ценим)

closed 2021-11-05 by Basoil