GyverLibs/microDS18B20 #8: Не работает при использовании GyverCore

ArtemiyKolobov, 2021-12-31:


GyverLibs, 2022-01-01:
Да, gyvercore слишком быстрая, тайминги уходят

ArtemiyKolobov, 2022-01-01:
Есть осциллограф, как будет время - поправлю тайминги для gyvercore.

ArtemiyKolobov, 2022-01-02:
Проблему локализовал: 
Неверно работает перезагрузка линии 1-Wire (функция _oneWire_reset(uint8_t pin)_ в файле **microOneWire.cpp**
Для исправной работы необходимо выждать между опусканием линии ведущим и замером напряжения линии 
```
...
13. delayMicroseconds(70);//Было 2, стало 70
...
```
Или
```
for (uint8_t c = 255; c; c--) {//Увеличим количество попыток считываний 80->>255
        if (!digitalRead(pin)) {
            #ifdef AVR
            SREG = oldSreg;//Зачем это делать здесь?? да ещё много раз?
            #endif
            for (uint8_t i = 255; !digitalRead(pin) && i; i--) {{//Увеличим количество попыток считываний 200->255
                delayMicroseconds(1);
            }
            return false;
        }
        delayMicroseconds(1);
    }
```
Работает и на GyverCore, и на родном ядре.

GyverLibs, 2022-01-08:
SREG = oldSreg.  
Безопасного включает прерывания, в отличие от sei(). И там не много раз, оно выполнится один раз перед выходом из функции

GyverLibs, 2022-01-10:
На всякий случай переделал на 255, хуже точно не будет, проверить на гуверкоре не было возможности

closed 2022-01-10 by GyverLibs