GyverLibs/microDS18B20 #12: Неправильные показания

k008, 2022-04-19:
Здравствуйте, на arduino nano при 2х контактном подключении, показывает 10.63 температуру (температура меняется только на ошибку), при обычном подключении 3.00 градусов и меняется при нагреве.

При этом датчик и соединение работает, так было проверено другой библиотекой
Подскажите как исправить?

GyverLibs, 2022-04-19:
Привет, что такое 2х контактное подключение?

k008, 2022-04-19:
Это когда ноги питания и земли датчика замкнуты и через резистор идут на питание к arduino, а налога данных подключается как обычно.
Его ещё называют "паразитное подключение"

GyverLibs, 2022-04-19:
А, ну вот паразитным оно и называется. В чем может быть проблема не знаю, у меня в тестах отлично работает на всех платформах. Возможно тайминги жёстче чем в других либах, но всё сделано чисто по даташиту

k008, 2022-04-19:
Куда смотреть?

GyverLibs, 2022-04-19:
Не работает пример или свой код? Если свой код, то есть ли в программе прерывания?

k008, 2022-04-19:
Не работает пример

GyverLibs, 2022-04-19:
Попробую смягчить тайминги, кину сюда файлик для теста ночью или уже завтра

GyverLibs, 2022-04-19:
Вообще это предельно странно, потому что библиотека проверяет корректность полученных данных. То есть датчик подтверждает, что температура 10.63

GyverLibs, 2022-04-19:
В файле oneWire.cpp в самом низу заменить функцию на эту
```
uint8_t oneWire_read(uint8_t pin) {
    uint8_t data = 0;
    for (uint8_t i = 8; i; i--) {
        data >>= 1;
        MOW_CLI();
        pinMode(pin, OUTPUT);
        delayMicroseconds(2);
        pinMode(pin, INPUT);
        delayMicroseconds(20);
        if (digitalRead(pin)) data |= (1 << 7);
        delayMicroseconds(40);
        MOW_SEI();
    }
    return data;
```

k008, 2022-04-25:
А у меня нет такой функции

GyverLibs, 2022-04-25:
Значит у тебя другая библиотека, не microds18b20

k008, 2022-04-25:
У меня библиотека http://www.pjrc.com/teensy/td_libs_OneWire.html

GyverLibs, 2022-04-25:
Эм, а зачем тогда писать мне в issue?..

k008, 2022-04-25:
В файле **oneWire.cpp** в самом низу заменить функцию на эту - Такой файл в чужой библиотеке, а вашей он называется, **microOneWire.cpp**. Я сразу про это не подумал.

Поправил в вашей библиотеке, показывает 2.88, вместо 23.12

GyverLibs, 2022-04-25:
При обычном подключении работает корректно?

GyverLibs, 2022-04-25:
Какой микроконтроллер используется? Какая версия библиотеки?

GyverLibs, 2022-04-25:
Режим паразитного питания не поддерживается библиотекой, могу попробовать добавить. Но сначала нужно разобраться почему не работает на обычном питании

k008, 2022-04-25:
Библиотека работает с 1820 или 18s20?

GyverLibs, 2022-04-25:
Библиотека только с теми, которые в названии - micro**ds18b20**

k008, 2022-04-26:
1. Я всё-таки другой датчик использовал (1820), вместо положенного 18b20
2. Паразитное питание не работает. Но подключение по 3-м проводам работает исправно.
3. Выше указанный кусок кода не пригодился, так как заработало всё со стандартными параметрами.
4. Использую arduino nano (atmega328p)
5. Для 1820 можно переделать библиотеку или проблематично?

GyverLibs, 2022-04-26:
Паразитное планирую добавить. Про 1820 никогда не слышал, но сделать можно все что угодно