GyverLibs/microDS18B20 #14: Не читается

thedemoncat, 2022-06-30:
```
// один датчик лучше читать без адресации, это сильно экономит память
#include <microDS18B20.h>
#define DC 5
MicroDS18B20<DC> sensor;
uint8_t address[8];

void setup() {
  Serial.begin(9600);
  sensor.setResolution(12);
}
void loop() {
  // запрос температуры  
  sensor.requestTemp();
    
  // проверяем успешность чтения и выводим
  if (sensor.readTemp()) 
    Serial.println(sensor.getTemp());
  else Serial.println("error");

  // Запрос адреса датчика
  if (sensor.readAddress(address)) {  // если успешно, выводим
    Serial.print('{');
    for (uint8_t i = 0; i < 8; i++) {
      Serial.print("0x");
      Serial.print(address[i], HEX);  // Выводим адрес
      if (i < 7) Serial.print(", ");
    }
    Serial.println('}');
  } else Serial.println("Not connected");

  // прпочитали сырое значение
  uint16_t rawVal = sensor.getRaw();  
  Serial.print("Сырые данные: ");
  Serial.println(rawVal);

  // преобразовали
  Serial.println(DS_rawToFloat(rawVal));

  // вместо delay используй таймер на millis(), пример async_read
  delay(1000);
}
```

GyverLibs, 2022-06-30:
Встроенные примеры работают корректно?

thedemoncat, 2022-07-01:
Только получение данных о температуре. Причем это единственная либа, с помощью которой я смог получить данные температуры. Остальные просто не видят датчик. Адрес вообще не могу считать ни одним из найденных способов(

GyverLibs, 2022-07-01:
- Датчик точно дс18б20? Существуют "аналоги", которые работают наполовину 
- Где покупался?
- Как подключен?
- Какая платформа (avr, esp)?
- Какая IDE?
- Версия IDE/компилятора/SDK?

thedemoncat, 2022-07-01:
1, 2. Датчик покупал этот - https://aliexpress.ru/item/4000550061662.html?gatewayAdapt=glo2rus&sku_id=10000002852197596&spm=a2g0s.12269583.0.0.6f254262JSG9FQ
3. Пробовал подключать проводами с резюком и вот [таким](https://aliexpress.ru/item/33050203027.html?item_id=33050203027&sku_id=12000023049024410&spm=a2g0o.search.i0.2.83fe9ff6XBLmiX) вот терминалом
4. Пробовал на 60-ти разных esp, D1mini - результат одинаковый
5,6.  PlatformIO Core, version 6.0.2. Но пробовал заливать и в штатной от ардуинки - результат одинковый

GyverLibs, 2022-07-01:
На этом терминале уже стоит резюк. Нужно либо резюк по схеме, либо терминал этот

GyverLibs, 2022-07-01:
Не исключен брак датчика. Температура 38.5 это правильная? Или погода на Марсе

GyverLibs, 2022-07-01:
Попробуй предыдущие версии библиотеки, например 3.7 и 3.5. у меня пока нет возможности проверить работоспособность на есп, но последний раз когда тестировал - все работало

thedemoncat, 2022-07-03:
Ок. Но через пару недель. Уехал в отпуск

thedemoncat, 2022-07-18:
Проверил работу с датчиками от другого вендора. все норм. Походу пришли чипы с непрошитым адресом(
Ишуз закрываю. Проблема решена

closed 2022-07-18 by thedemoncat