GyverLibs/microDS18B20 #15: Не читает датчики с новой версией библиотеки Arduino ESP32 Version 2.0.3

voluser, 2022-06-30:
До этого стояла версия платы 2.0.2  , после обновления на 2,03  перстала работать библиотека microDS18B20. Моя плата ESP32-DevKitC V4 модуль ESP32-WROOM-32D. В менеджере плат выбран модуль ESP32 Dev module. Пробовал опрашивать датчики через библиотеки OneWire + DallasTemperature  - все отлично . Только после отката версии платы на Arduino ESP32 Version 2.0.2 , заработала и microDS18B20 .

GyverLibs, 2022-06-30:
И снова индусы что то сломали или переделали в IO, как же это надоело...

DenysChuhlib, 2022-09-25:
У меня тоже самое с переходом с esp8266 на esp32 не работает.

DenysChuhlib, 2022-09-25:
Это люди отправляли свои ришения проблем и кто-то накосячил

GyverLibs, 2022-09-25:
А?

DenysChuhlib, 2022-09-25:
Если глянуть в историю исправлений то там куча людей что-то пофиксили.

GyverLibs, 2022-09-25:
Это были Фиксики. В любом случае, если оно перестало работать - IO в esp32 стал работать как минимум на 100-300 микросекунд медленнее, это п.дец) пока нет возможности, а так планирую залезть в эту либу с лог анализатором и нормально всё руками потрогать

DenysChuhlib, 2022-09-25:
Хорошо

PcbSmd, 2023-03-03:
Насколько я понимаю, пока нет решения? Модуль ESP32-WROOM-32E, PlatformIO IDE v3.0.0. Библиотека microDS18B20 не видит датчик, но, как сказано выше - OneWire + DallasTemperature работают.

DenysChuhlib, 2023-03-03:
Да

GyverLibs, 2023-03-03:
да толку фиксить, всё равно сломают) ждём стабильного ядра

AntonMalygin, 2023-04-28:
> да толку фиксить, всё равно сломают) ждём стабильного ядра

Блин я думал я что-то не так напаял...действительно az-delivery-devkit-v4 на данный момент 28апр.23г пока не работает библиотека. Жаль..Так приятно использовать ваши библиотеки в своих проектах.

GyverLibs, 2023-04-28:
DS18B20 планируется крупное обновление, расширение функционала, но пока неизвестно когда к сожалению. С esp32 проблемы были изначально, они несколько раз решались но с новыми обновлениями ядра опять всё ломается. Несмотря на то, что библиотека написана невероятно примитивно - протокол OneWire реализован на digitalWrite/Read и задержках, что должно работать на любой Ардуино-совместимой платформе

sigmashig, 2023-05-03:
Не заработало на  ESP32. Посмотрел реализацию paulstoffregen... Или он намудрил или на тривиальных задержках, в условиях многоядерности, уже не получится написать. У него постоянные отключения/включения ISR. Возможно, эти преывания и мешают - сбивают задержки.

GyverLibs, 2023-05-03:
Прелести RTOS :)

uzi18, 2023-06-22:
For ESP32 use OneWireNg, it is rewritten from scratch.

closed 2024-01-25 by GyverLibs