GyverLibs/microDS18B20 #17: Определение наличия датчика на шине

brightproject, 2022-12-03:
Когда используют "голый" код без библиотек, то используют такого вида конструкцию:
```
    if ( !ds.search(addr))
    {
      //Serial.println("1-wire scan ended.");
      ds.reset_search();
      delay(250);
      SensorFound = false;
      //return;
    }
```
Каким образом определить наличие или отвал датчика в случае с использованием библиотеки OneWire?

GyverLibs, 2022-12-03:
Читать документацию к библиотеке OneWire

brightproject, 2022-12-03:
> 

В вашей библиотеке

> microDS18B20.h

Не нашел, как определяется наличие или обрыв датчика на линии.

brightproject, 2022-12-03:
Что за переменная idx?
Это просто сквозная нумерация датчиков на линии - 1,2,3 и т.д.?

GyverLibs, 2022-12-03:
Номер адреса в массиве. Как в примере https://github.com/GyverLibs/microDS18B20/blob/main/examples/async_read_many/async_read_many.ino

brightproject, 2022-12-03:
> Номер адреса в массиве. Как в примере https://github.com/GyverLibs/microDS18B20/blob/main/examples/async_read_many/async_read_many.ino

В чем принципиальное отличие от [async_read_many_bus] и [async_read_many_bus_pgm]?

GyverLibs, 2022-12-03:
Ну как бы в использовании PROGMEM для хранения адресов..

brightproject, 2022-12-03:
> Ну как бы в использовании PROGMEM для хранения адресов..
С PROGMEM  понятно, а отличие async_read_many_bus от async_read_many просто в коде или по скорости работы есть отличия или в количестве одновременных сенсоров на шине?

GyverLibs, 2022-12-03:
Режим progmem даёт возможность хранить адреса в progmem памяти, преимущество в этом

brightproject, 2022-12-04:
> Режим progmem даёт возможность хранить адреса в progmem памяти, преимущество только в этом, других отличий нет

С этим понятно, спасибо за пояснения.
Подскажите пожалуйста, у меня на шине 2 датчика, адреса очень разные, похоже из разных партий датчики.
Читаю температуру с обоих датчиков так:
`sensor[i].getTemp();`
i в цикле перебирает два сенсора, дело в том, что один датчик выдает информацию с точностью до десяты и кратность 0.5
А второй датчик выдает точность до сотой, с кратностью 0.01
Что это и как два датчика из разных партий заставить выводить с одинаковой градацией?
Задавать setResolution?

GyverLibs, 2022-12-04:
Не очень понял про кратность и точность. Количество знаков после запятой зависит от того, как вывести значение. Если обе температуры выводятся одинаковым способом в цикле - то да, копать в сторону установки одинакового разрешения

GyverLibs, 2022-12-04:
Это зависит от китайцев и конкретного подвала, в котором сделан датчик. Разрешение оригинальных датчиков 12 бит по умолчанию

closed 2024-01-25 by GyverLibs