GyverLibs/microDS18B20 #23: Не Проблема, но вопрос )

vadlis, 2023-10-04:
Просто интересно - что мешает добавить флаг в readTemp и по этому флагу вызывать requestTemp в концу функции? тогда код примеров будет покороче )

GyverLibs, 2023-10-04:
Ничего не понял

vadlis, 2023-10-04:
float getTemp(uint8_t idx = 0, bool request = false) {
 ...
 if (request) {
  requestTemp(idx);
  }
}

тогда в примерах можно убрать код с requestTemp.

GyverLibs, 2023-10-04:
Это нарушит логику работы с датчиком. Температуру нужно запросить, а ответ будет готов спустя некоторое время вплоть до 1 секунды

vadlis, 2023-10-04:
понятно

vadlis, 2023-10-04:
хотя не очень )). код из примера:

https://github.com/GyverLibs/microDS18B20/blob/main/examples/async_read_many/async_read_many.ino#L35-L45

в этом случае requestTemp просто переместится из скетча в библиотеку.

GyverLibs, 2023-10-04:
Сделай форк с более короткой записью. В данной библиотеке request это отдельная операция на усмотрение юзера

vadlis, 2023-10-04:
Ну собсно флаг тоже был бы опциональным. Ну да ладно - на нет и суда нет ))

closed 2023-10-04 by vadlis